pyros-dev/rostful #12: Python unit tests

stonier, 2015-04-20:
Good idea to get familiar with how catkin packages do unit testing

* python unit tests (simple python api unit test scripts)
* ros unit tests (i.e. fire up several python nodes and get them to talk to each other)

Perhaps we can cover this on wednesday, and I can add this to the 'yujin dev' notes I'm writing as we do them.

asmodehn, 2015-04-29:
**Tests**

Catkin has tools that let you build/run python nosetests for simple python api testing, rostests for testing communication between nodes and rocon tests for testing communication between multimasters. capabilities has nose and ros tests. gatways have nose and rocon tests. I really ought to put down some easy docs somewhere for our team on how to build/run these. Perhaps we can do this in an hour session next week.

asmodehn, 2015-04-29:
I think the code is too unstable yet to start putting tests around it...

stonier, 2015-04-30:
Aha..the programmers conundrum. 

Which comes first...the program or the tests?  :neckbeard:.

Whatever way helps you manage this is ok by me.

asmodehn, 2015-04-30:
Hehe, it s more like an infinite loop of specification -> prototype -> test
-> finalize -> specification -> ...

And writing test for, and finalizing, things for which specifications will
change doubles the workload for little benefit at the beginning, while
specification is not stable enough.

My point for working on admin part right now also... code structure will
change significantly.

stonier, 2015-05-02:
Yeah, I am typically of the same view.

Though I have seen some use cases where the permutations of inputs to a black box are large and this makes it important to automagically test that the system is working against these so that your team mates are not annoyed by your broken code, even if it is in a development phase.

We're not at that point with this though.

asmodehn, 2015-12-17:
Integration REST test started. test framework still a bit buggy.
Flask Unit Test structure in place, but no test implemented yet.

asmodehn, 2016-02-11:
Needs unittests before 0.1.0 to make sure we dont break anything during cleaning up.