pyros-dev/rostful #4: Wrap head around rocon interactions/android remocon/web remocon

stonier, 2015-04-20:
This is currently what we use for human2robot and human2system. It is dependent on rosbridge and zeroconf.

* Interactions are what tells the user what he can run.
* Remocons are the user's application to connect/start web apps
* Android remocon can do 'discovery' and nfc.

* @jihoonl can help you with the overall running of interactions and the web remocon.
* @dwlee/@stonier can show you the android remocon.

stonier, 2015-04-20:
@jihoonl, @dwlee please post some links here

jihoonl, 2015-04-20:
ROS Wiki : http://wiki.ros.org/rocon_interactions
Github : https://github.com/robotics-in-concert/rocon_tools/tree/indigo/rocon_interactions
Tutorials: http://wiki.ros.org/rocon_interactions/Tutorials

Demo : I will demonstrate on 2015.04.22 afternoon

asmodehn, 2015-04-24:
So from what I understood : 
Step 1 (soon) : We just need to advertise ROSTful as an Interaction.
Step 2 (later) : Rostful needs to be able to list the interactions that are authorised/forbidden for a user and the topics/services/actions/etc. used by that interaction. This information will be used to implement authorisation mechanisms.

asmodehn, 2015-04-24:
There is also the fact that remocon cannot always be accessed remotely : 
It needs to be deployed onto the robot, and accessible at a well known URL.
From there it can be used to access ROSTful.

asmodehn, 2015-04-28:
Realized today that what rostful needs is actually the interface of a Rapp : 

publishers: 
  - name: chatter
    type: std_msgs/String
subscribers: []
services: []
action_clients: []
action_servers: []

From this information we already know what to expose or not as rest. The client part of the Rapp will communicate through these.
The interactions is just a way to advertize which Rapp is available to who.

asmodehn, 2015-04-28:
Rostful now advertise itself as a webapp interaction 8284e44acaeac99fd543152128eec3383c648a72.
Tested with http://webapp.robotconcert.org/rocon_web_remocon/

asmodehn, 2016-02-11:
Dropping this as rocon features have been dropped from Pyros https://github.com/asmodehn/pyros/issues/61 and we try to KISS here.
Anyway all rocon features are implementable on top of bare ROS REST API provided by rostful if needed.

closed 2016-02-11 by asmodehn