pyros-dev/rostful #6: Plan for serving pages

stonier, 2015-04-20:
Do after getting head around interactions.

asmodehn, 2015-04-20:
Relevant topics and documentation : 

* Separate multiple webapps / WSGI servers : 
Via URL, via subdomains ? via different WSGI servers (port numbers).
Idea : using http://werkzeug.pocoo.org/docs/0.10/middlewares/
URL seems safer / more flexible for our usecase anyway. (AlexV)

* If we have a long running service online, how to do versioning  ( URL, etc . ) ? 
http://stackoverflow.com/questions/389169/best-practices-for-api-versioning
There are a lot of threads about this. just google.
=> do we need to care about this now ? or maybe not ?

* Production vs Dev :
Using a proper WSGI server for production ( not werkzeug run_simple method which should be used only for development ) : http://werkzeug.pocoo.org/docs/0.10/deployment/proxying/#configuring-nginx
Other options : http://werkzeug.pocoo.org/docs/0.10/deployment/
The main point is to not bring in the whole Flask dependency if we can avoid it. KISS.

asmodehn, 2015-04-24:
ROSTful is a webserver delivering dynamic pages + REST API.
No versioning system for now ( maybe we will never need it )
Flask is in already.
HTTP server will also be used ( gunicorn / tornado ? ). still trying to KISS.

asmodehn, 2015-12-17:
Serving with flask for development, with tornado on robot.
Basic jquery mobile UI pages for easy access to the REST API.
No plan for versioning API yet. still quite early...
Closing.

closed 2015-12-17 by asmodehn