pyros-dev/rostful #8: Unit test launchers for rest2ros action implementations

stonier, 2015-04-20:


asmodehn, 2015-04-28:
Done, with template pages to send Goal easily. 5037e93f992d3af45d59e32ed3cfab8ac8b209d3

Unexpected things encountered ( with turtle_actionlib ) : 
* Starting as a node from a launchfile, the client node is named like the action 'turtle_shape', no matter the name I pass to it in the launchfile. Confusing since the action is 'provided' by the server, not the client.
* In RQT all action_topics are grouped under one 'topic display'. Not sure which is sub which is pub for which node. no relation shown between client and server topics.
* It seems the topic list report all actions topics for both server and client, which can be confusing ( how to know which is which ?, if you have one server and 2 clients, how can we discover which action we need to use from outside ? -> might need more info ). -> Pointless to send a goal to a topic named from an action client, but how can we know by ROS introspection ?
* The Goal message to send has to have a specific structure ( that I found out by looking into the source code. ) this is not obvious without a client lib to handle it -> pbm to port a client using actionlib to multiple languages.

closed 2015-04-29 by asmodehn

stonier, 2015-04-29:
> Starting as a node from a launchfile, the client node is named like the action 'turtle_shape', no matter the name I pass to it in the launchfile. Confusing since the action is 'provided' by the server, not the client.

Action servers and clients are named - it's only because the tutorial passed the node name in as the action name that makes this confusing. Typically changing the node name doesn't affect anything.

> In RQT all action_topics are grouped under one 'topic display'. Not sure which is sub which is pub for which node. no relation shown between client and server topics.

`rqt_graph` is a bit of a shambles still. rqt superceded the old wxwidget tools they had, but nobody's given it any real love. I just noticed that alot of the functions there even don't work (including the `hide actions` - it didn't even show me the pubs and subs when that was off. Don't pay too much attention to this tool :neckbeard:

> It seems the topic list report all actions topics for both server and client, which can be confusing ( how to know which is which ?, if you have one server and 2 clients, how can we discover which [...]

I have code in the rocon gateways ([here](https://github.com/robotics-in-concert/rocon_multimaster/blob/indigo/rocon_gateway/src/rocon_gateway/master_api.py#L159) which extract action clients and servers from a running system. Basically you go direct to the master (or via some of the python api to the master), get a full list and parse it to work out what is what. If you need this functionality, let's pull that code into somewhere common that you can reuse (make a new issue here and I'll do that job).

> The Goal message to send has to have a specific structure (that I found out by looking into the source code. ) this is not obvious without a client lib to handle it -> pbm to port a client using actionlib to multiple languages.

Yes, this is a problem point for non cpp/python communications. No full actionlib support. Java had a dodgy one for a while, and you can workaround to some extent by just directly talking to the pubs & subs without a client library. If you think this non-blocking, long duration style service request is going to be a very common pattern for us, we need to make an early decision to go with the pre-empting actionlib, or extend the pubsub pairs.

asmodehn, 2015-04-29:
About actions and names, to be precise, my launch file was : 
```
  <node pkg="turtle_actionlib" name="turtle_shape_server" type="shape_server"/>
  <node pkg="turtle_actionlib" name="turtle_shape_client" type="shape_client"/>
```

and rostopic gives : 
```
/turtle_shape/cancel
/turtle_shape/feedback
/turtle_shape/goal
/turtle_shape/result
/turtle_shape/status
/turtle_shape_server/cancel
/turtle_shape_server/feedback
/turtle_shape_server/goal
/turtle_shape_server/result
/turtle_shape_server/status
```

But I was expecting either : 
```
/turtle_shape_client/cancel
/turtle_shape_client/feedback
/turtle_shape_client/goal
/turtle_shape_client/result
/turtle_shape_client/status
/turtle_shape_server/cancel
/turtle_shape_server/feedback
/turtle_shape_server/goal
/turtle_shape_server/result
/turtle_shape_server/status
```
OR only : 
```
/turtle_shape/cancel
/turtle_shape/feedback
/turtle_shape/goal
/turtle_shape/result
/turtle_shape/status
```

About pulling action list, it s interesting to know it s doable. Not sure yet if we ll need it or not...

The code needed for actionlib seems to be very small so far, so yes it s probably non-blocking.
I ll try to get the pubsubs pairs soon.