pyros-dev/rostful #14: Sync message_conversion module with other libraries

jihoonl, 2015-04-21:
rostful uses [forked version of json message conversion](https://github.com/asmodehn/rostful/blob/master/src/rostful/message_conversion.py) from [rosbridge_library](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py).

It would be better to use shared library instead of keeping the fork. Three options

* Use [rosbridge_library message_converter](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py)
* Use [rospy_message_converter](https://github.com/baalexander/rospy_message_converter)
* converge rosbridge_library version of message converter and rospy_message_converter, then use it in rostful.

Note that rospy_message_conveter may not have resolved [this issue](https://github.com/RobotWebTools/rosbridge_suite/issues/93).

asmodehn, 2015-04-28:
Actually I stumbled upon a json message generation problem today.
For actions, I need to generate a goal object from the webpage doing the REST request.
But I am not using any js side library yet since every message i needed to send so far was very simple.
I did it by copying the code for now : 51445f0bcbcb9bf5ad640ebc19f774016ea940aa

So now I am wondering which ros js library I should have on the client side that will allow me to build ros json structures. I do not need the whole roslib.js, since I don’t need to do any websocket connections, etc. I need just the part to build json messages for ROS.

It would be the exact inverse of the json message conversion used in the rostful backend... in javascript, in the (web) client.

asmodehn, 2015-06-19:
Transferring this issue to https://github.com/asmodehn/rostful-node/issues/6