pyros-dev/rostful #15: Investigate how to do Authentication/authorization with WSGI apps

asmodehn, 2015-04-21:
http://wsgi.readthedocs.org/en/latest/specifications/simple_authentication.html

But in case of emergency : Another potential candidate for WSGI libs : [Bottle](http://bottlepy.org/docs/dev/index.html#)...

asmodehn, 2015-04-21:
http://en.wikipedia.org/wiki/Comparison_of_web_application_frameworks#Python_2

stonier, 2015-04-22:
Talked to the guys here - what we will do for field testing is just do security at the wireless connection level so people can't hack the robots.

Full user/admin application login & authorisation can be postponed till after the field testing.

asmodehn, 2015-04-22:
Few considerations regarding Deployment, Authentication, and Authorization : 
* ROSTful includes a web-server (gunicorn/tornado) that can serve the WSGI python app. We can copy how rosbridge_server integrates with tornado. This server opens a specific port and wait for Web Requests.
* If there is a rocon_interactions node present that can provide Rostful with the list of interactions, and the topics associated with this interactions, Rostful can also provide the list of interaction via a rest service. It can also determine the authorizations to connect to an interaction, via the authorization mechanism already used in rocon_interactions. An interdiction to connect to an interaction will probably implies the interdiction to connect to any of the exposed topics/services/actions for this interaction...
* Authentication should be setup and maintained on the rocon system. However it should be possible to authenticate, even if we connect to a robot, and this robot cannot communicate with the rocon system. The robot will probably keep a memory of the latest password database he got from the rocon system.
* Authentication can be configured via web interface, and should be usable from a standard web server ( as well as from ROS/rocon )
* Remocon will have to be updated to be able to connect via REST ( and be able to do everything without using rosbridge and websockets. Webapps/Interactions could however use websockets if they need to )
* Rocon_interactions should be updated to be able to deliver the list of topics/services/actions they are using. This is needed to infer the authorization rules for a user for each of the ROS topics/services/actions.

asmodehn, 2015-12-17:
To keep rostful simple, authentication should be done in another place , for the web traffic.
For robot traffic we would need to rely on a Robot wide security, which doesnt really exist yet.

The solution for now is to lock the robot down as much as possible ( minus a few ports open for things to work ), and provide authentication at the web level only, while rostful will support token based authentication with an external backend, and provide a very basic authorization mechanism.

An interesting project for this external authentication service, that we should investigate, is https://github.com/LukeB42/microauth