pyros-dev/rostful #16: Investigate how rostful may or may not be deployed.

asmodehn, 2015-04-21:
Since rostful contains a WSGI app ( and or maybe should contain a full web server ? ) deployment is an issue that need to be tought about early on.

Starting point : https://realpython.com/blog/python/kickstarting-flask-on-ubuntu-setup-and-deployment/
=> w need to match this with ROS usual deployment processes.

stonier, 2015-04-22:
Ah...that might be tricky. Let's talk about that/release process this afternoon.

asmodehn, 2015-12-17:
Rostful starts his own web server, we cannot really avoid it.
However we should strive to have only one web server per robot.
Example : 
- if someone needs rosbridge to hook onto -> integrate the WSGI app it in rostful
- if someone needs a webapp, just provide a html file

For simplicity we should also avoid database migration and other complex setup when possible. We need to keep the configuration minimal, and with simple systems that just work ( ie. a launch file somewhere ).

The question that remains is where/how do we manage the (deployment specific) configuration file...

Note configuration might be related to the robot, more than the software, and rostful could make use of another "robot_config" package...