pyros-dev/rostful #17: Include a HTTP server into rostful

asmodehn, 2015-04-24:
We can follow how rosbridge_server uses tornado to serve websockets.

asmodehn, 2015-04-24:
@jihoonl I realised that rosbridge actually embeds tornado https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosbridge_server/src/tornado instead of using it on the OS as a dependency. Correct ?

Any specific reason for this ? Wouldn't it work if we just list it as a dependency and use it from there ?

For rostful I would prefer if I could keep on the OS what can be installed by dependency system... It would help keeping the package small and easily maintainable I think...

jihoonl, 2015-04-24:
I remembers that the reason was to use the most recent version of tornado. The deb version of tornado was too old to use in Trusty. 

python-tornado version  : 3.1.1
rosbridge : 4.0.2

asmodehn, 2015-04-28:
I made a version with gunicorn 7858340e64603a83e6730b79be8327a92bda317f. Ubuntu package is 0.17, which needs to be launched via shell. Not ideal.
Version 0.19 allows for starting from python script which would be better : http://docs.gunicorn.org/en/latest/custom.html

asmodehn, 2015-06-15:
Closing this as we are using the flask server during development. creating #34 for checking gunicorn implementation.

closed 2015-06-15 by asmodehn