pyros-dev/rostful #20: Release Processes (sticky)

stonier, 2015-04-27:
We have a few means of releasing software

**System Debs**

If they're already on ubuntu, simply a matter of adding them to the [rosdep database](https://github.com/ros/rosdistro/tree/master/rosdep). Regular debs into `base.yaml` and python debs into `python.yaml`.

**Catkin Debs**

Just regular catkin open source packages that can be rolled on the osrf build farm. The software that does this is `bloom`.

Some notes on how to do this at [yujin_release](https://github.com/yujinrobot-release/yujin_release/wiki). There are references back to more detailed pages there. 

Can also build the debs locally if you want to test them.

**Catkin 3rd Party Debs**

This lets you build 3rd party debs from other people's sources (so you don't have to fork/maintain yourself). It also uses bloom. Notes at the same page as above. Can also build the debs locally if you want to test them.

Catkin 3rd party debs go into `/opt/ros/indigo`. Touching system files is awkward, but there is some conversation from Mike Purvis about this on the mailing lists (just google).

**Yujin Debs**

For custom rolling debs of our own. This is useful for things that would effect system directories (e.g. /etc) that catkin 3rd party debs dont. I use CPack to do this but can also be a custom build. Ping me (dan) if you need to do this.

**Python Pips**

Rosdep used to install these way back. Haven't seen anything to do this after the new catkin build system came out, but it may still be possible. Need to google it.

asmodehn, 2015-04-29:
We might want to keep this in mind : https://github.com/ros/rosdistro/pull/8227#issuecomment-97268112

stonier, 2015-04-29:
Got it :neckbeard:

asmodehn, 2015-12-17:
The goal here is to release rostful first on yujin buildbot, with the minimum set of dependencies ( released as third party ). So we should try to avoid too many dependencies.
This is still an ongoing task.