pyros-dev/rostful #22: Split topics into publishers / subscribers

asmodehn, 2015-04-28:
The idea is to match the Rapp interface specification design.

asmodehn, 2015-06-03:
Why should we ? seems going backward to directional connection paradigm. Topic paradigm is supposed to make things easier, isnt it ?

stonier, 2015-06-05:
For rapps it's about introspection. A rapp claiming it has a `sensor_msgs/Image` topic doesn't tell me whether it's a camera streaming images for me, or an image processing node that consumes images. For programs, having publisher/subscriber declarations lets it automatically validate connections to a certain degree that would otherwise require a human to contextually interpret from the name of the topic.

Graphs are alot richer when the edges have directional information than none.

asmodehn, 2015-06-05:
Yes for rapps it makes sense.

But for rostful launch/configure parameter maybe it doesnt : a user wants
to expose a topic/action, he probably doesnt want to care if it s pub or
sub or doesnt want to think if he wants to pub or sub to it.
And maybe he change his mind halfway and doesnt want to reconfigure rostful
to suddenly publish to it instead of just listening...

asmodehn, 2015-12-17:
Actually talked about this again recently.
We will probably do it soon since it seems to make more sense for ROS people.
The main argument being that we usually want to expose Publisher OR Subscriber to a topic, but rarely both. @ stonier : correct ?