pyros-dev/rostful #27: Review

stonier, 2015-04-29:
**Pre-Catkin**

```
import roslib
roslib.load_manifest('rostful')
```

This is a hangover from the pre-catkin build system. No longer necessary in catkin.

**Templates**

If you want to make sure the templates get installed with the python package, then you need to make sure they get added to your `setup.py`. A more complicated example with package data is [here](https://github.com/yujinrobot/yujin_tools/blob/master/setup.py). You can test that everything gets installed and works by doing the install step - either `yujin_make --install` or if you are just using catkin_make, then build as normal and then run `make install` from the build directory. Open a new shell (with no ROS env. variables), switch to `your_workspace_dir/install` (should be next to `devel`) and `source setup.bash`. You can then test your installation here and this exactly matches the deb installed package.

**Tests**

Catkin has tools that let you build/run `python nosetests` for simple python api testing, `rostests` for testing communication between nodes and `rocon tests` for testing communication between multimasters. [capabilities](https://github.com/osrf/capabilities/tree/master/test) has nose and ros tests. [gatways](https://github.com/robotics-in-concert/rocon_multimaster/tree/indigo/rocon_gateway_tests/tests) have nose and rocon tests. I really ought to put down some easy docs somewhere for our team on how to build/run these. Perhaps we can do this in an hour session next week.

**Documentation**

I see you've already started with the sphinx infra. Das ist good! The OSRF build farm will automatically build and host these docs. These are then accessed from the 'Code Api' links on the ros wiki (e.g. the [code api](http://docs.ros.org/indigo/api/rocon_uri/html/) for [rocon_uri](http://wiki.ros.org/rocon_uri). For python its becoming my preferred way to do open source catkin package documentation - and in the near future they're going to find a way to have this style of documentation supercede everything that is on the wiki (e.g. distro based tutorials). I typically only doc new or fixed things though - already too much code out there.

**Code**

Looks fine to me, but I don't have any sense of the entire design in a quick parse. Is there anything in particular you'd like me to take a close look at?

asmodehn, 2015-04-29:
**Templates**
Ah true I never tested the installed software... I ll make an issue about that to keep it in mind.

**Tests**
Didnt make any tests so far... I ll make an issue about that to keep it in mind.

**Documentation**
Didnt write any doc yet... I ll make yet another issue about that to keep it in mind.

**Code**
Nothing in particular. it s still changing a lot and I haven’t been able to decide on a final design yet.

We might want to start using milestone to prioritize these issues...

asmodehn, 2015-12-17:
Rostful keep on moving, this issue stayed behind. Main points have been addressed, tests and docs are coming soon. closing.

closed 2015-12-17 by asmodehn