pyros-dev/rostful #32: Killing rostful when run from roslaunch nicely

asmodehn, 2015-06-03:
Killing rostful is problematic.
On python 2.7 running using debug=True creates a child process ( used to reload the code when changed ) but killing it leaves the child ( web server ) alive. Might be fixed with python 3.X
reference : 

when running with debug = False , killing the node triggers : 
^C[rostful_server-1] killing on exit
[rostful_server-1] escalating to SIGTERM

=> We want to be able to kill it nicely with SIGINT, without the need to escalate to sigterm -> something in flask / roslaunch / rostful preventing it to finish gracefully ?

asmodehn, 2015-06-03:
After quick investigation, we should either :
* disable rospy signal handlers so that flask KeyboardInterrupt can be catched as in any python  program
* extract the flask server part outside of a rosnode, and connect to ros services -> split rostful in two parts : rostful_web / rostful_node

stonier, 2015-06-03:
Does Flask not listen for SIGINT/SIGHUP?

asmodehn, 2015-06-03:
Flask, when running in debug mode has a complex behavior ( launching a
child process that respawn ).

Python itself has some signal handler that trigger keyboardinterrupt
exception on sigint.

Rospy also seems to have its own signal handlers.

I disabled the debug feature and it seems to work good enough for now.
On the long term we should probably extract flask from the ros package.
This problem will disappear.

But we ll need to think about deploying a configured setup of all our
software... flask requires DB and config files presetup if we dont want to
spend hours setting this up on each robot.
-> container ?

asmodehn, 2015-06-15:
Closing this, as what could be done right now has been done.

closed 2015-06-15 by asmodehn