pyros-dev/rostful #41: autogenerated documentation

asmodehn, 2015-07-30:
People in the web world have no idea about ROS. and looking through all the .msg / .srv files is far from ideal.

It would be nice, since the REST API is automatically generated from ROS, to also generate some documentation along with it, explaining all the parameters, etc.

Maybe find a way to translate/make available the ros documentation into some kind of REST API autogenerated doc...

asmodehn, 2015-08-11:
how about that : http://www.flaskapi.org/ ?

stonier, 2015-08-12:
I think this is something we should approach at a more fundamental level, not a rostful specific level.

With many changes happening around 2.0 I wouldn't mind approaching this topic with the core ROS community itself.

asmodehn, 2015-08-13:
I think there are 2 issues :
- ros documentation ( messages and whats in it )
- rest documentation ( what a get request does, what a post request does,
etc. )
I would be surprised if ros communications match rest exactly, since they
have different purposes.

This issue is only about generating the rest part of it. It could reference
the ros message one to make things simpler where possible.

asmodehn, 2015-12-17:
We currently have a beginning of python sphinx docs.
For REST interface, the clearer and most uptodate is probably the Rester json test files at the moment.
We should try to find a better way for autodocuenting REST APIs...