pyros-dev/rostful #45: callbacks sample for ros topics

vallabh0, 2015-08-07:
I have a ros program that publishes data on to a particular topic in ros frequently. I am developing a web based application(JS) using REST to communicate to ros through ROSTFUL on the server end.

What i need is to be able to specify a callback function through the web app which gets called whenever the data in the topic is updated.

I have already tried this with rosbridge . It works fine there but i need my web based client to be using restful apis.

I have come across some provisions already made for this purpose in rostful-node repository but i require a sample describing how to call those functions.

For example passing a callback function to the turtlesim for position which gets called whenever the position data gets updated and my web app always has the latest data without refreshing the page.

asmodehn, 2015-08-07:
If you want to be using a restful API, then I think you need to query the url regularly ( fast enough for your human user ) to get updates from your ROS system, in that case from your topic.

REST works with a ( relatively slow - but fast enough for human ) polling system, where ros works with a ( relatively fast ) pushing system. The main reason ~I think~ is that the polling paradigm (client server) handles better network accidents, like network going down, IP changes, etc.
So I don't think the use case you are thinking about is a REST use-case ( more like a ROS habit that you are trying to apply to the web world).

Even if you get a callback in the server, your webpage will need to poll that data to display it.  The server will not be able to send you that data unless you request it from your browser ( or unless we duplicate what rosbridge is doing with web sockets, which is not the point here )

An advanced setup could be that there is a REST URL that returns you some javascript code.
Then you run that code into a webworker, and it set ups for you a listener ( using roslib.js in webpage and rosbridge in the backend ).
I also would like to use a similar system to display the ROS log in the rostful webpages, but rostful and rostful_node are not able to do that just yet.
It s on my todo list, but the priority is quite low, so I am not sure when I ll get time to work on that.

In the mean time I really think that you mean to call a REST service in a loop from your javascript, and use that data...

If you think I am mistaken and there is really a proper REST way to do this, please feel free to let me know.

asmodehn, 2015-12-17:
Javascript sample code/pages would definitely be good in documentation...
Maybe even a demo webpage if we can find the proper incantations to run the python code in a browser to simulate ROS there...