pyros-dev/rostful #49: indigo.rosinstall trying to checkout to non existing indigo branch of Rostful.

dhirajdhule, 2015-09-04:
https://github.com/asmodehn/rostful/tree/indigo-devel/rostful_examples 
instructions on this page took me to install using indigo.rosinstall, which tries to checkout to indigo branch but there is only indigo-devel branch for rostful.

Its not causing any issue except error while wstool init.

asmodehn, 2015-09-11:
Thank you for reporting this.
The indigo branch was probably lost at some point while refactoring to interface with ros via another process. I honestly cannot remember.
Anyway things have stabilized a bit now so I (re)created an indigo branch, which I believe should solve this problem.
Please let me know if you encounter any other issue while running rostful.
I ll keep this issue open for a little while, just in case.

dhirajdhule, 2015-09-11:
Thanks.
There's one more issue.
The turtlesim_dev.launch in
roslaunch rostful turtlesim_dev.launch --screen
is not in rostful but in rostful-examples, in indigo-dev branch.
If you have corrected it in new indigo branch then there is no issue.
Anyway rostful was working fine when I used indigo-dev.rosinstall.

asmodehn, 2015-09-11:
Fixed readme : 
https://github.com/asmodehn/rostful/commit/89c33e1b3e97e2c43175cd8b28ad7f85e71f910b

closed 2015-11-24 by dhirajdhule