pyros-dev/rostful #69: Tornado crash

bit-pirate, 2016-01-29:
When running `rocon_gocart`:

```
ERROR:tornado.application:Uncaught exception
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tornado/http1connection.py", line 234, in _read_message
    delegate.finish()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tornado/httpserver.py", line 280, in finish
    self.server.request_callback(self.request)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tornado/wsgi.py", line 280, in __call__
    WSGIContainer.environ(request), start_response)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1836, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1820, in wsgi_app
    response = self.make_response(self.handle_exception(e))
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
    return original_handler(e)
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/flask_cors/extension.py", line 188, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1403, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
    return original_handler(e)
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/flask_cors/extension.py", line 188, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/lib/python2.7/dist-packages/flask/views.py", line 84, in view
    return self.dispatch_request(*args, **kwargs)
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/flask_restful/__init__.py", line 579, in dispatch_request
    resp = meth(*args, **kwargs)
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/rostful/flask_views.py", line 270, in get
    services = self.node_client.services()
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/pyros/pyros_client.py", line 219, in services
    res = self.services_svc.call(send_timeout=5000, recv_timeout=10000)  # Need to be generous on timeout in case we are starting up multiprocesses
  File "/opt/yujin/amd64/indigo-stable/lib/python2.7/dist-packages/zmp/service.py", line 123, in call
    raise ServiceCallTimeout("Did not receive response through ZMQ socket.")
ServiceCallTimeout: Did not receive response through ZMQ socket.
```

asmodehn, 2016-01-29:
This happens on the robot startup using rocon_gocart.launch

asmodehn, 2016-02-11:
We need to implement more solid timeouts + exception handling all the way from ROS to the web.
work in progress there : https://github.com/asmodehn/rostful/tree/removing_rocon_features