pyros-dev/rostful #77: Can't install and use rostful

Kenairod, 2016-06-27:
Hello, it's been two days since I've been trying to use rostful, but I can't make here is the series of commands I've been running in order to avoid all the error messages I encountered, but it's still not enough :
`sudo apt-get install python-pip
sudo apt-get install virtualenvwrapper
echo "source /etc/bash_completion.d/virtualenvwrapper" >> ~/.bashrc

cd ~/catkin_ws/src
git clone --recursive https://github.com/asmodehn/rostful.git

mkvirtualenv rostfulenv --no-site-packages
workon rostfulenv

cd ~/catkin_ws/src/rostful
pip install catkin-pkg

cd ~/catkin_ws/src/rostful/deps
git clone --recursive https://github.com/asmodehn/pyros.git
cd ~/catkin_ws/src/rostful/deps/pyros
pip install .
cd ~/catkin_ws/src/rostful/
pip install .

pip install mock
pip install simplejson
pip install tornado`

But still when I try to run :

asmodehn, 2016-07-07:
Hello,

Thank for taking the time to try it and report this issue.
From your environment it looks like you are mixing catkin workspaces with python venv, which is one of the target for rostful. However this is tricky to get right for all usecases, so thanks again for the details you put here.

First, it's likely that the doc you had a look at is outdated.
Feel free to make a PR, or otherwise point me in the right direction and I ll try to find some time to fix it.

```
python -m rostful flask
```
Is an outdated command.

If you check https://github.com/asmodehn/rostful/blob/indigo-devel/rostful/__main__.py you ll see the available python commands and arguments.

So this _should_ work out of the box with indigo-devel version
```
python -m rostful 
```

About
```
roslaunch rostful rostful.launch
```
It looks like pyros is failing to import some modules : 
```
WARNING:root:Error: Could not import PyrosROS from .rosinterface.pyros_ros cannot import name core
```
We would need to dig deep to find where that "core" module is imported. but that is likely caused by an unexpected python setup.

Running `python -m site` would give you an idea of the different path where python is looking for modules, and when jumping between a catkin workspace and a virtualenv, it is likely that we find some issues.

In your commands I don't see anywhere `source devel/setup.bash` which is the usual catkin way to get the PYTHONPATH modified to include ros library (after you build your workspace with catkin_make). it might be just that or it might be something more complicated...

I haven't investigated your usecase fully yet (but I am very interested in doing it - just didnt get the time, since I use primarily the catkin workflow), so if you want something to run quickly I would advise, for now, to stick to the catkin+rosdep way of doing things.

And if you have some motivation to help and  dive into how to use python and virtualenvs with catkin workspaces, I can point you to https://github.com/asmodehn/catkin_pip and more specifically to https://github.com/asmodehn/pyros/tree/config_refactor where the latest development is going on.

asmodehn, 2016-07-07:
Ah Another useful piece of information...
This : 
```
WARNING:root: => Emulating ROS setup now for distro indigo and workspaces ()
WARNING:root:Appending path /opt/ros/indigo/stacks to ROS_PACKAGE_PATH
WARNING:root:Appending path /opt/ros/indigo/share to ROS_PACKAGE_PATH
WARNING:root:Appending path /opt/ros/indigo/bin to PATH
WARNING:root:Appending path /opt/ros/indigo/lib to LD_LIBRARY_PATH
WARNING:root:Appending path /opt/ros/indigo/lib/pkgconfig to PKG_CONFIG_PATH
WARNING:root:Prepending path /opt/ros/indigo/lib/python2.7/dist-packages to PYTHONPATH
WARNING:root: => ROS setup emulation done.
```
is created by https://github.com/asmodehn/pyros-setup.

This is the package in charge of loading the ROS environment, when someone uses python directly, without sourcing the ROS environment as per the normal catkin workflow.
I understand this is your case...

There are two versions currently out there of this package : 
- 0.0.8, installed from ros packages, you might get it from rosdep. This one is old and maybe doesn't work anymore (I haven't tested it for a while). I will need to get it deleted from ros repositories as soon as all dependent packages have moved to use >=0.1.0 version
- 0.1.0 most recent released version available at https://pypi.python.org/pypi/pyros_setup
I recently added a config file to specify which path need to be added to the python path, and the docs might be outdated at the moment...

*If you go for using the python workflow*, you want to make sure you have the latest pyros_setup from pypi installed, and you want to make sure the config file points to your ros workspaces (built catkin workspace, including any underlays). Config file should be created if it doesn't exist...

*If you go for catkin workflow*, the pyros-setup is not needed and should not even be imported (since 0.1.0)

I know all this is tricky, so feel free to ask any questions. It will also help get these python packages working properly with ROS.

Cheers !

asmodehn, 2016-07-15:
Just to let you know I recently made a detailed tutorial to use rostful the catkin way : 
https://github.com/asmodehn/rostful/blob/indigo-devel/doc/tutorial.rst

Hopefully this should help you get it running.