pyros-dev/rostful #91: Not picking up topics in one hit

stonier, 2016-10-20:
To reproduce, simply run the gopher rocon podium. On launchup you will see it picking up topics...one by one.

```
[INFO] [WallTime: 1476941046.219288] /rostful Pyros.rosinterface : Creating rosinterface topic /sentience/reality Reality
[INFO] [WallTime: 1476941047.233487] /rostful Pyros.rosinterface : Creating rosinterface topic /core/buttons/stop Empty
[INFO] [WallTime: 1476941048.243496] /rostful Pyros.rosinterface : Creating rosinterface topic /sentience/profile Profile
```

asmodehn, 2016-10-24:
I found out more details about this.
Actually the algorithm picks up topics in one hit.
But each topic interface needs some time to establish connection (pub and sub), and currently I have a loop with a 1 second sleep in there. More details at https://github.com/asmodehn/pyros/issues/104

asmodehn, 2017-04-06:
@yotabits this seems to have been fixed in pyros a while ago. Can you confirm please ?