pyros-dev/rostful #92: Local subscriptions causing it to blow up

stonier, 2016-10-20:
To reproduce (100%):

* Run gopher rocon on podium
* Open a podium konsole
* Run `rostopic echo /sentience/reality`
* CTRL-C the rostopic echo

Over in the main window:

```
received. 0 bytes were received. Please check sender for additional details.
[INFO] [WallTime: 1476941171.550023] /rostful Pyros.rosinterface : Deleting rosinterface topic /sentience/reality Reality
[WARN] [WallTime: 1476941171.550878] Could not process inbound connection: Internal error processing topic [/sentience/reality]{'message_definition': "##########################################################################################\n#\n# For exposing the robot's perception of itself and the world.\n#\n##########################################################################################\n\n##############################\n# Status Indicators\n##############################\n\nuint8 STATUS_IDLE                             = 0\nuint8 STATUS_BUSY_PARKING                     = 1\nuint8 STATUS_BUSY_UNPARKING                   = 2\nuint8 STATUS_BUSY_DOCKING                     = 3\nuint8 STATUS_BUSY_UNDOCKING                   = 4\nuint8 STATUS_BUSY_TRAVELLING                  = 5\nuint8 STATUS_BUSY_WILL_DEPART                 = 8\nuint8 STATUS_BUSY_CANCELLING                  = 9\nuint8 STATUS_BUSY_OTHER                       = 10\nuint8 STATUS_WAITING_FOR_AUTHENTICATION       = 50\nuint8 STATUS_WAITING_FOR_INPUT                = 51\nuint8 STATUS_PAUSED_EMERGENCY                 = 100\nuint8 STATUS_BROKEN_OTHER                     = 200\n\n##############################\n# Status Messages\n##############################\n\nstring STATUS_IDLE_MSG                        = bored, waiting for something to do\nstring STATUS_BUSY_PARKING_MSG                = finding a place to park/dock\nstring STATUS_BUSY_UNPARKING_MSG              = unparking/undocking\nstring STATUS_BUSY_TRAVELLING_MSG             = travelling between locations\nstring STATUS_WAITING_FOR_AUTHENTICATION_MSG  = waiting for a hooman to authenticate\nstring STATUS_WAITING_FOR_INPUT_MSG           = yawn, waiting for input\nstring STATUS_BUSY_WILL_DEPART_MSG            = received a new job and preparing to move out\nstring STATUS_BUSY_CANCELLING_MSG             = current job cancelled so heading home\nstring STATUS_BUSY_OTHER_MSG                  = engaged in some nefarious activity and do not wish to share\nstring STATUS_PAUSED_EMERGENCY_MSG            = a rogue human has activated the emergency button\nstring STATUS_BROKEN_OTHER_MSG                = broken, urgently needing help from a hooman\n\n##############################\n# Other Flags\n##############################\n\nint8 CONCERT_NEVER_CONNECTED=-1\n\n##############################\n# Other Messages\n##############################\n\nstring MESSAGE_BABBLE=Is babbling incoherently, you're going to need a babelfish\nstring MESSAGE_FEELINGS=Feelings are dumb, and should be hated\nstring MESSAGE_SAD=I dont have emotio
...
...
```

asmodehn, 2016-10-25:
This should be fixed by https://github.com/asmodehn/pyros/pull/103/commits/29821f8577ba229009b3d8b2edcc39cb2ae66252

asmodehn, 2016-10-25:
There was a simple logic issue when determining which topic interface should be dropped or not.
Not sure how much this is related to our problem with the gateway, and if that fix will have any impact.

stonier, 2016-10-26:
Note...there is no gateway in a gopher_rocon launch. As discussed, the multimaster subscriber problem is a different thing entirely.

asmodehn, 2017-04-06:
@yotabits this should have been fixed a while ago from pyros. Can you confirm the fix please ? If the problem cannot be reproduced I will close it.