pyros-dev/rostful #144: flask error

rvipin17, 2018-10-11:
if I do roslaunch rostful rostful.launch, getting following error: 

`[rostful-2] process has died [pid 9754, exit code 1, cmd /home/addverb/catkin_mir_ws/src/rostful/scripts/rostful -h 127.0.0.1 -p 8080 -s flask -c /home/addverb/catkin_mir_ws/src/rostful/instance/rostful.cfg -l /home/addverb/catkin_mir_ws/src/rostful/instance/rostful.log ~connections_list:=/connection_cache/list ~connections_diff:=/connection_cache/diff __name:=rostful __log:=/home/addverb/.ros/log/9dec2b6a-cd19-11e8-996d-dca26629e16b/rostful-2.log].
log file: /home/addverb/.ros/log/9dec2b6a-cd19-11e8-996d-dca26629e16b/rostful-2*.log
`

rvipin17, 2018-10-11:
**Rostful install:** 

Navigate to your workspace,
$ cd ~/catkin_ws/src 
$ git clone -b gopher-devel --recursive https://github.com/pyros-dev/rostful.git
$ cd rostful
$ cd deps
$ git clone --recursive https://github.com/pyros-dev/pyros-utils.git
$ cd pyros-utils
$ pip install .
$ cd ..
$ git clone --recursive https://github.com/pyros-dev/pyros.git
$ cd pyros
$ pip install .
$ cd ../..
$ pip install .
$ pip install -r requirements.txt
$ cd ../..
$ catkin_make
$ source devel/setup.bash
$ roslaunch rostful rostful.launch

**The above steps will solve all dependencies with rostful. Browse to localhost:8080, you’ll be able to see your rostful localhost running.** 

> Correct me if I am wrong

asmodehn, 2018-10-11:
So there are a few different ways to do this...
- You could clone everything ( even all the ros source ) and build
everything together.
- You could download packages from ros instead for things you are not
interested in modifying/rebuilding.
- You could run directly from a python setup with a virtualenv, and use
pyros-setup to discover your ros distro dynamically on runtime.

Or any combination of these. But you need to understand how your system
works, how ros setup.bash modify your environment, how python expect his
environment, and how pyros-setup can find ros distro by modifying your
python runtime, and not your whole system (like ros does).

asmodehn, 2019-06-19:
Hi,
In case it might help, I have updated a rostful usecase with turtlesim : https://github.com/asmodehn/rostful-examples/pull/6
Let me know if you find it useful.
Thanks.