pyros-dev/rostful #163: Unable to Launch rostful

arwinder-sesto, 2019-02-11:
$ roslaunch rostful rostful.launch
[rostful.launch] is neither a launch file in package [rostful] nor is [rostful] a launch file name
The traceback for the exception was written to the log file

I dont see the launc folder or the launch files for rostful. Did i miss something out?

asmodehn, 2019-02-11:
This is a python package. It should be launched as any python package : `python -m rostful`.

That being said, this is not "a simple ros node". It s a web rest framework(multiprocesses) that happens to interface with ROS using pyros. But mostly, it is a python rest backend.

If you havent use a python web framework yet, I advise you get familiar with one first. FWIW I'm using flask in rostful.

arwinder-sesto, 2019-02-11:
Thank you. 

I will look into this. Sorry I am kind of new to this.

asmodehn, 2019-06-19:
Hi,
In case it might help, I have updated a rostful usecase with turtlesim : https://github.com/asmodehn/rostful-examples/pull/6
Let me know if you find it useful.
Thanks.