pyros-dev/rostful #177: Compatible with dynamic reconfigure

goodahn, 2019-05-02:
Currently, it's not able to change parameter that is created by dynamic reconfigure.
Although I added UI part for updating parameter value, it can't be done.
Because of a type conflict and missing member, rostful can't update parameter.
Could you resolve this problem?

asmodehn, 2019-05-03:
Hello, thanks for reporting this issue.
It used to work, from what I can remember, so lets investigate...

I have a few questions : 
- Which version are you using ? latest from master ?
- Where is the error exactly ? It seems to me at first glance that a dynamic reconfigure related error would not appear in rostful, since rostful is only a WSGI frontend for ROS. So I would imagine the error would appear lower, in one of the pyros packages... but I might be wrong.
- Where is the code ? Can you show me what you have done somewhere so I can replicate the issue?
Any [MCVE](https://gamedev.meta.stackexchange.com/questions/2201/whats-a-minimum-complete-verifiable-example-mcve-and-how-do-i-provide-it) ?

Finally, this software is free, which means it is also yours, so feel free to fork and change what you need to make it work, an send a PR ;-)

Let me know if you find more clues !

goodahn, 2019-05-03:
Thank you for your kind answer!

- I use rostful 0.2.1 version which is the latest from master.

- This is my simple [MCVE](https://github.com/woouTW/rostful/commit/bc5ac2738d584c3959a0f8dbed08ca48eb569a12).
(It was implemented without considering PR.)

Please let me know if there are any problems :)

asmodehn, 2019-05-08:
Oh I see what you are doing there :-)
You create a ROS node directly in the flask [WSGI](https://en.wikipedia.org/wiki/Web_Server_Gateway_Interface) app.

Actually this was originally the design of rostful :  a simple wsgi app that is also a ROS node.
You can have a look at the [original version from BenKehoe](https://github.com/benkehoe/rostful)

Although its simplicity is a strength, there was problems with this design : 
- Environment Setup : all web code now also depends on ROS, our web code must run in a ROS environment (and ROS messes up your PYTHONPATH)
- Security: Database code injection is already known, when robot code injection will become famous ?
- Maintenance : All our python dependencies must now be ported to ROS, and in web there is a lot...
- Performance : Scaling our webserver potentially impact our ROS system. and most webserver scale automatically, based on request arriving from potentially everywhere...

So I had to accept this design was too simple, and something had to be done, even if it ended up a bit more complex.

Thats where [Pyros](https://github.com/pyros-dev/pyros) comes from.
With PyROS design, there is a strict separation of concern (relying on OS separation of processes) : 
- pyros deals with pure python things only.
- interfaces are somewhat 'pluggable' in that framework (most notably [pyros_interfaces_ros](https://github.com/pyros-dev/pyros-rosinterface)) to deal with specific systems.
- rostful becomes just a normal python flask app, running with Pyros taking care of ROS stuff.

So I advise that you have a look at the code and track the 'get_pyros_client()' call.
You will see it retrieves an object that is previously set with [set_pyros_client](https://github.com/pyros-dev/rostful/blob/master/rostful/context.py#L11) from [Server.launch](https://github.com/pyros-dev/rostful/blob/master/rostful/server.py#L113)

It is around that time that pyros is launched (another process !), and pyros is creating the actual ROS node. (you can check it on rqt).
You can see it there : https://github.com/pyros-dev/pyros-rosinterface/blob/master/pyros_interfaces_ros/pyros_ros.py#L34

So you ll need to dive deeper into the code (don't forget to take your debugger along for the ride).
Having multiple processes makes it harder to debug, but gives a strong isolation to address all the issues I talked about before.

You should be able to add code for dynamic reconfigure into the original pyros node...
There used to be one a long time ago.
Eventually I decided to move it out because : 
-  there are many ways to configure the node (file, launch params, etc.), and it is tricky to test them all, some being dynamic, and others not, especially when there are multiple process involved...
- there is https://github.com/awesomebytes/ddynamic_reconfigure that might be simpler to integrate (I haven't tried)
- and I wasn't using it :)

So feel free to play around and send a PR when you get something working, but more importantly, I think pyros needs tests, and we should try to identify where things can be made simpler, but not too simple...

The next piece of the puzzle for me is the launch workflow : how these processes are launched? how can they find each other ? how can each find their configuration ? quite a tricky problem...

goodahn, 2019-05-10:
Thank you for your kind explanation!
As you said, I already experienced such dependency problems...
I will look into about this problem more and if I get a good solution, I will send a PR.
Thank you!

asmodehn, 2019-06-19:
Hi,
In case it might help, I have updated a rostful usecase with turtlesim : https://github.com/asmodehn/rostful-examples/pull/6
Let me know if you find it useful.
Thanks.

goodahn, 2019-06-24:
Thank you for letting me know this update :)
And sorry for reading your answer late.
Thank you!