pyros-dev/rostful #184: ROS service exit with timeout

michaelblindnology, 2021-07-21:
Hi,
I'm using **ROS1 melodic**, branch **gopher-devel**.
I'm using rostful for ROS service.
My service takes 2-3 seconds.
The ROS service returns successfully, but rostful returns with **500**.
```
An exception occurred! => 500 
{'traceback': {'tb_frame': {'f_code': {'co_name': 'post', 'co_filename': 'rostful/api_0_1/flask_views.py'}, 'f_globals': {'__name__': 'rostful.api_0_1.flask_views', '__file__': 'rostful/api_0_1/flask_views.pyc'}}, 'tb_next': None, 'tb_lineno': 355}, 'exc_type': "<type 'exceptions.NameError'>", 'exc_value': "global name 'PyrosServiceTimeout' is not defined"}
ERROR:rostful:An exception occurred! => 500 
{'traceback': {'tb_frame': {'f_code': {'co_name': 'post', 'co_filename': 'rostful/api_0_1/flask_views.py'}, 'f_globals': {'__name__': 'rostful.api_0_1.flask_views', '__file__': 'rostful/api_0_1/flask_views.pyc'}}, 'tb_next': None, 'tb_lineno': 355}, 'exc_type': "<type 'exceptions.NameError'>", 'exc_value': "global name 'PyrosServiceTimeout' is not defined"}
```
If I not call my service, but return true, everything is working.
Could please assist,

Thanks Michael

asmodehn, 2021-07-24:
Hi,

I had a quick look and tried to jog my memory... hopefully this will give you some clues.

Looking at the code I guessing you made a call here: https://github.com/pyros-dev/rostful/blob/gopher-devel/rostful/api_0_1/flask_views.py#L314
Then accessing that part of the code : https://github.com/pyros-dev/rostful/blob/gopher-devel/rostful/api_0_1/flask_views.py#L351 your interpreter seem to not know the `PyrosServiceTimeout` global name.
This one is dynamically imported here : https://github.com/pyros-dev/rostful/blob/gopher-devel/rostful/api_0_1/flask_views.py#L107
Not sure why it would not be defined, but it doesn't hurt to review your virtualenvironment setup, how your run your interpreter, etc.

In any case, attach a debugger to your server, put some breakpoints, and you will be able to track what exactly happens. 

It is possible this behaviour is triggered by some untested code path underneath, but the error itself only says the symbol is not defined... so it is the first thing to investigate.

asmodehn, 2021-07-24:
I noticed also there is a pull request impacting that part of the code for gopher-devel : https://github.com/pyros-dev/rostful/pull/108/files You should probably try it out.

Let me know if that works for you, I ll merge it.

michaelblindnology, 2021-07-25:
Hi,
My service respond took > 5 seconds, so I exit on timeout.
I changed `/home/ubuntu/.local/lib/python2.7/site-packages/pyzmp/service.py#90 `
in function ` def call(self, args=None, kwargs=None, node=None, send_timeout=10000, recv_timeout=20000, zmq_ctx=None):`
`recv_timeout` from 5000 to 2000.It fixed the problem.
How can I change  recv_timeout from flack_views ?
Is it possible to run my ROS services in diffent flask threads/process ?
E.g make_plan of move_base in thread1, other service in thread 2 ?
THanks in advance,

asmodehn, 2021-07-25:
Hi,

Oh good so you already started digging inside the code ;-)

From flask_views, the pyros client code is calling the service, you ll have
to follow the call path and check where to pass that timeout, I wont have
time to dig deeper at the moment.

pyzmp is a multiprocess python library i wrote to make multiprocess
communication easier(from my perspective) than what was available at the
time. It uses zeroMQ (via pyzmq) for interprocess&network communication.

Communications between ros nodes (already processes) and python processes
happens there, so I m not sure what you mean since everything is already
multiprocess by default...

Rostful is a web layer on top of all that, but to simplify things, you
should be able to do communicate with ROS just with pyros and a python
repl... although that might take some getting used to to setup everything
correctly.

Good luck, let me know how it goes ;-)
I might be able to jump on a chat session sometime next month if I can be
of some help.

Cheers,
--
AlexV